okadan/flutter-nfc-manager #231: NDEF tag does not contain any NDEF message

channdara, 2025-05-19:
Hi. I'm running into an issue with iOS cannot detect tag. I'm working with Mifare NTAG (ISO 14443) which can detect fine in Android platform but not iOS. There are no error in Flutter log at all but when I open Xcode, I received this error below:

**Error Domain=NFCError Code=403 "NDEF tag does not contain any NDEF message" UserInfo={NSLocalizedDescription=NDEF tag does not contain any NDEF message}**

I did some research and found a fix that required me to write a record to the TAG, whatever TEXT or URL or somethings. And yes, the issue gone and iOS platform can detect TAG normally. But the problem is, I cannot manually add record or tell user to add a record to all the TAGs. Is there any way to solve this issue? Am I doing something wrong or missing any iOS configuration? Please let me know, thank you so much.

### My configuration on iOS side

- I have added **NFCReaderUsageDescription** to info.plist
```
<key>NFCReaderUsageDescription</key>
    <string>This app uses NFC to scan and interact with compatible tags for enhanced functionality and a seamless user experience.</string>
```
- I have added **Near Field Communication Tag Reader Session Formats Entitlement**
```
<key>com.apple.developer.nfc.readersession.formats</key>
<array>
    <string>TAG</string>
</array>
```
- I have added these to info.plist too
```
<key>com.apple.developer.nfc.readersession.felica.systemcodes</key>
<array>
    <string>12FC</string>
    <string>0000</string>
</array>
<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
<array>
    <string>D2760000850100</string>
    <string>D2760000850101</string>
    <string>D2760001180101</string>
    <string>00000000000000</string>
</array>
```
- I have added **Near Field Communication Tag Reading** under **Signing & Compatibilities** in **Xcode**

channdara, 2025-05-19:
I have fork the repo and make a small change on error condition in **NfcManagerPlugin** at line 507
```
if let error = error as? NSError, error.code != 403 {
```

I'm not a Swift guy and I know checking like this is not the correct way. Just want to point out there. If you have better solution please do so. Thank you.

channdara, 2025-06-18:
@Priyanshu-Bej 

I have a temporary fix in the forked repo here: https://github.com/channdara/flutter-nfc-manager
Can be use for workaround for now, still waiting for the main repo to update this.

komaxx, 2025-07-03:
For everyone wondering how you can use @channdara's excellent fix (thank you!), add this to your pubsec.yaml:

```yaml
dependency_overrides:
  # Override as `nfc_manager` currently does not scan tags with empty NDEF blocks on iOS
  # TODO: Check if still required for nfc_manager versions > 4.0.2
  nfc_manager:
    git:
      url: https://github.com/channdara/flutter-nfc-manager.git
      path: packages/nfc_manager
      commit: d5480bc
```

smplfy, 2025-07-22:
Any word on official fix being released. As appreciated as it is to have an option, I hesitate to use @channdara's fix in my production app. Thanks @okadan