okadan/flutter-nfc-manager #236: Reader not working on iOS with package 4.x

philitell, 2025-07-02:
I've used this package successfully in an production app. Recently i updated from 3.5.1 to 4.0.2 and nfc scanning stopped working for ios. On Android everything works fine. On iOS however no tags are recognized at all.
After starting a session the bottom sheet (system overlay) opens and nothing happens when trying to scan a tag.
I havn't changed any ios specific configurations
 - it works with version 3.5.1
 - no system reaction with version 4.0.2

Are there any configuration specific for ios and package version 4.X that i'm missing (and that are not in the Readme)

philitell, 2025-07-02:
Is this related to #233 ?

komaxx, 2025-07-03:
I have the same issue.
For me, I can see in XCode that the tag is actually read but it runs into this issue:

```
Error Domain=NFCError Code=403 "NDEF tag does not contain any NDEF message" UserInfo={NSLocalizedDescription=NDEF tag does not contain any NDEF message}
```

So, my understanding is that many NFC Tags come pre-formatted from the factory formatted with some empty NDEF block already on it.
With 3.x.x versions of the lib that was ok, with 4.0.x iOS disregards such tags. For me, I can scan the same tag on iOS if I just write some random string on it using an Android phone.

I'll borrow a bit into the native code to see what's different...

philitell, 2025-07-09:
I my case i does not make any difference if the tag is empty or not. When i start an ios-reading session i see this error in Xcode: 
`
[ERROR:flutter/shell/common/shell.cc(1064)] The 'dev.flutter.pigeon.nfc_manager.FlutterApiPigeon.tagSessionDidBecomeActive' channel sent a message from native to Flutter on a non-platform thread. Platform channel messages must be sent on the platform thread. Failure to do so may result in data loss or crashes, and must be fixed in the plugin or application code creating that channel.
`

philitell, 2025-07-09:
@komaxx Do you have any issues when calling startSession on iOS?

dominikIncibit, 2025-09-22:
same issue here. Any solution?